mcci-catena/Catena-Sketches #4: Need provisioning step-by-step guide

terrillmoore, 2017-01-15:
Provisioning requires:

1. adding the CPUID to DevEUI mapping in `MCCI-Catena4410-Arduino-Library/src/lib/gk_WellKnownCpuBindings.cpp` (and coincidentally defining the base platform to be used)
2. adding the keys to TTN using the website or the CLI
3. adding the device-specific LoRaWAN keys to `catena-lorawan-provisioning/extra/gk_lorawan_keys.project.cpp`

There needs to be a better step-by-step guide for this.

forrestfiller, 2017-08-06:
@terrillmoore some dissonence between the updated readme and information provided here. For example, I was unable to complete the LoRaWAN provisioning section successfully--couldn't get my device registered to my application. However, once I found this issue, specifically the "Reprovisioning" section at the end of this issue, I entered in 
```
lorawan configure nwkskey 0
lorawan configure appskey 0
lorawan configure devaddr 0
lorawan configure netid 0
```
And then the serial monitor went into a loop of EV_TXSTART, EV_TXSTART, etc.

Then I git cloned [your random sysEUI generator here](https://github.com/things-nyc/random-eui64) into my project dir, ran:
```
cc random-eui.c
./a.out
``` 
and grabbed the generated sysEUI printed in my terminal window. You reference it as a potential devEUI above, so on the TTN console I was able to change the devEUI there, and use the command from the above:
```
system configure syseui yoursyseui
```
in my serial monitor in atom via platformIO (which could be done in any terminal with cli installed):
```
pio device monitor --port /dev/cu.usbmodem1421 --baud 115200
```
then I ran through the [LoRaWAN Provisioning commands from the README.md](https://github.com/mcci-catena/Catena4410-Sketches/blob/master/catena4450m101_sensor/README.md#lorawan-provisioning) using the generated sysEUI as my devEUI, and now I'm finally seeing packets hitting my backend on TTN console. 

This is a bit of a debug dump (sorry). I'm make a pull on your README, and hopefully you can cherry pick somethings that are useful to help people out.

Thanks for your support, excited to have the Catena4450 up and running via TTN.

forrestfiller, 2017-08-06:
Another quick thing to consider adding into the README.md is the **payload decoder** option in the TTN console. 

I dropped your code from your [Catena message decoder for TTN](https://github.com/mcci-catena/Catena4410-Sketches/blob/master/extra/catena-message-0x14-0x15-decoder-ttn.js) into my decoder function on the TTN console and now I have lovely human readable data to view without converting hex in my brain - which I can't do thaaaaat well at this point anyway.

terrillmoore, 2017-08-19:
Hm, the info here is better than some of the info in the README.md. Leaving this open in hopes that I'll remember to update the README. But we need to split up this repo (keep 4410 separate from 4450, etc.)