mcci-catena/Catena-Sketches #5: Document how to set up sleep mode for the SAMD BSP for Adafruit

terrillmoore, 2017-01-15:
It's a bit tricky to update the Adafruit BSP with the required sleep-mode changes. Please document.

1) need the locations on Windows, Mac, Linux.
2) Need the instructions:
   ```bash
   % cd {somewhere}/arduino15/packages/adafruit/hardware/samd/1.0.13
   % git remote add mcci https://github.com/mcci-catena/ArduinoCore-samd.git
   % git pull mcci
   ```

forrestfiller, 2017-08-06:
@terrillmoore your [README.md](https://github.com/mcci-catena/Catena4410-Sketches/blob/master/catena4450m101_sensor/README.md) does a nice job of walking people through this process, you may be able to close this issue at this point: [SAMD BSP instructions](https://github.com/mcci-catena/Catena4410-Sketches/tree/master/catena4450m101_sensor#patch-the-adafruit-bsp-package)

terrillmoore, 2017-08-19:
Thanks @forrestfiller -- fixed by 7702e64efa8d88362983884603f5f9444e27ae0c

closed 2017-08-19 by terrillmoore