mcci-catena/Catena-Sketches #6: Issues faced while build the code

svelmurugan92, 2017-05-24:
While build the Catena4450m101_Sensor faced some minor issues. Listed it below.

1. gk provision key not found
Issue :
`Arduino: 1.8.2 (Windows 7), Board: "Adafruit Feather M0"

C:\Users\velmurugan\Documents\Arduino\libraries\Catena4410-Arduino-Library\src\lib\Catena4410_gk_lorawan_keys.cpp:120:90: fatal error: ../../../catena-lorawan-provisioning/extra/gk_lorawan_keys.project.cpp: No such file or directory

         #include "../../../catena-lorawan-provisioning/extra/gk_lorawan_keys.project.cpp"

                                                                                          ^
compilation terminated.`
My changes: 
Looks like these are needed for 4410 not for 4450. 
So, I Remove/rename Catena4420_gk_lorawan_keys.cpp and Catena4410_gk_lorawan_keys.cpp. 
To prevent it from compiling. (I know, There is some better way to do that).

2.  adjust_millis_forward function not declared.
Issue : 
`Arduino: 1.8.2 (Windows 7), Board: "Adafruit Feather M0"

E:\ThingsNetwork\Things-Network-Chennai\source\MCCI Catena\Catena4410-Sketches-master (1)\Catena4410-Sketches-master\catena4450m101_sensor\catena4450m101_sensor.ino: In function 'void settleDoneCb(osjob_t*)':

catena4450m101_sensor:551: error: 'adjust_millis_forward' was not declared in this scope

     adjust_millis_forward(CATCFG_T_INTERVAL  * 1000);

                                                    ^
exit status 1
'adjust_millis_forward' was not declared in this scope`

My changes: 
I think, Code not able to link proper delay.h file.
So, I copied the adjust_millis_forward function definition before the function call. 
I took adjust_millis_forward function definition from file C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino\delay.c

terrillmoore, 2017-05-26:
This is really two issues. I will split this up.

terrillmoore, 2017-05-26:
See issue https://github.com/mcci-catena/Catena4410-Arduino-Library/issues/1 -- this is really a problem with that library.

terrillmoore, 2017-08-19:
Closed by https://github.com/mcci-catena/Catena4410-Arduino-Library/commit/cb2a0f4cf1b6287010d90954ad61aebd4a21e53b

closed 2017-08-19 by terrillmoore