mcci-catena/Catena-Sketches #7: `adjust_millis_forward` function not declared.

terrillmoore, 2017-05-26:
From @svelmurugan92:
 
adjust_millis_forward function not declared.
Issue : 

> Arduino: 1.8.2 (Windows 7), Board: "Adafruit Feather M0"
> 
> E:\ThingsNetwork\Things-Network-Chennai\source\MCCI Catena\Catena4410-Sketches-master (1)\Catena4410-Sketches-master\catena4450m101_sensor\catena4450m101_sensor.ino: In function 'void settleDoneCb(osjob_t*)':
> 
> catena4450m101_sensor:551: error: 'adjust_millis_forward' was not declared in this scope
>
>    adjust_millis_forward(CATCFG_T_INTERVAL  * 1000);
>                                                     ^
> exit status 1
> adjust_millis_forward' was not declared in this scope

My changes: 
I think, Code not able to link proper delay.h file.
So, I copied the adjust_millis_forward function definition before the function call. 
I took adjust_millis_forward function definition from file C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino\delay.c

terrillmoore, 2017-05-26:
Hm. I wonder what is going on.  I don't find that code in my tree. In any case, I created `adjust_millis_forward` for the MCCI version of the Adafruit platform (see https://github.com/mcci-catena/ArduinoCore-samd). As part of the current build, I recommend installing that version of the BSP. Can you please research where `adjust_millis_forward()` came from? Something seems wrong. It seems unlikely that they would have chosen a name identical to the one I chose. I don't see that I pushed this upstream...

Also note that you're using an ARM BSP, so the AVR definition (while useful) is not the right one; can you check to confirm the exact function and compare to the code in our repository in https://github.com/mcci-catena/ArduinoCore-samd/commit/d16834145d7965cd31b0513cc4661f69edc5804b?

terrillmoore, 2017-07-31:
See instructions for updating BSP in catena4450m101_sensor's [README.md](https://github.com/mcci-catena/Catena4410-Sketches/blob/master/catena4450m101_sensor/README.md).

svelmurugan92, 2017-08-01:
Hello Terry, 
Followed the steps as mentioned in the  [README.md. ](https://github.com/mcci-catena/Catena4410-Sketches/blob/master/catena4450m101_sensor/README.md)

At the end step, Not able to rebase the master. It returns as below 
```
$ git rebase master
Cannot rebase: You have unstaged changes.
Please commit or stash them.
```
Please suggest.

terrillmoore, 2017-08-01:
Say “git status” and examine what you have.

 

If following instructions on a clean system, you would have no unstaged changes. (git checkout followed by git rebase would leave no modified files – unless the area was already modified before you started.)

 

So you must use `git status` to tell you what is changed, and then work out how to deal with it. The good news is, that anything that is changed is either the result of a compile (hence disposable) or something you changed locally for test (and also disposable or stashable).

 

Probably `git stash` followed by `git rebase` will do what you want, but it depends on what you’ve done. But since you were working with this previously, and (I know from reports) had odd problems, I think.

 

Also, I think that two other people tried the instructions tonight without difficulty.

 

--Terry

terrillmoore, 2017-08-19:
Remaining issues fixed by 7702e64efa8d88362983884603f5f9444e27ae0c

closed 2017-08-19 by terrillmoore