mcci-catena/Catena-Sketches #40: Serial.dtr

trlafleur, 2018-07-27:
You are using this non-standard function in your code, but it's NOT in your libraries??? Nor is it documented in your getting started pages...

where can it be found?? is it from Paul's Stoffregen Teensyduino code base??

Thanks

terrillmoore, 2018-07-27:
It's in the Adafruit BSP for the feather m0, and therefore in our BSP as well. What BSP are you using?

trlafleur, 2018-07-28:
I see that now, was using the default M0 board BSP...

I built a separate directory and library to make sure I was getting just your stuff... but it still looks like I'm missing something...

As a note: you may want to add a reference to YOUR Fram library as its different from the current adafruit version.

thanks

Arduino: 1.8.5 (Mac OS X), TD: 1.42, Board: "Adafruit Feather M0"

~~~

Linking everything together...
"/Users/lafleur/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-gcc"  "-L/var/folders/53/f8mlg0xn1xv3fsm58phqmtb40000gn/T/arduino_build_966129" -Os -Wl,--gc-sections -save-temps  "-T/Users/lafleur/Library/Arduino15/packages/adafruit/hardware/samd/1.2.1/variants/feather_m0/linker_scripts/gcc/flash_with_bootloader.ld" "-Wl,-Map,/var/folders/53/f8mlg0xn1xv3fsm58phqmtb40000gn/T/arduino_build_966129/catena_hello_lora.ino.map" --specs=nano.specs --specs=nosys.specs -mcpu=cortex-m0plus -mthumb -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -o "/var/folders/53/f8mlg0xn1xv3fsm58phqmtb40000gn/T/arduino_build_966129/catena_hello_lora.ino.elf" "/var/folders/53/f8mlg0xn1xv3fsm58phqmtb40000gn/T/arduino_build_966129/sketch/catena_hello_lora.ino.cpp.o" "/var/folders/53/f8mlg0xn1xv3fsm58phqmtb40000gn/T/arduino_build_966129/libraries/Catena-Arduino-Platform/Catena-Arduino-Platform.a" "/var/folders/53/f8mlg0xn1xv3fsm58phqmtb40000gn/T/arduino_build_966129/libraries/Catena-mcciadk/lib/mcciadklib_buffertoulong.c.o" "/var/folders/53/f8mlg0xn1xv3fsm58phqmtb40000gn/T/arduino_build_966129/libraries/Catena-mcciadk/lib/mcciadklib_formatdumpline.c.o" "/var/folders/53/f8mlg0xn1xv3fsm58phqmtb40000gn/T/arduino_build_966129/libraries/Catena-mcciadk/lib/mcciadklib_multiszindex.c.o" "/var/folders/53/f8mlg0xn1xv3fsm58phqmtb40000gn/T/arduino_build_966129/libraries/Catena-mcciadk/lib/mcciadklib_safecopystring.c.o" "/var/folders/53/f8mlg0xn1xv3fsm58phqmtb40000gn/T/arduino_build_966129/libraries/Catena-mcciadk/lib/mcciadklib_snprintf.c.o" "/var/folders/53/f8mlg0xn1xv3fsm58phqmtb40000gn/T/arduino_build_966129/libraries/Catena-mcciadk/lib/mcciadklib_stringcomparecaseinsensitive.c.o" "/var/folders/53/f8mlg0xn1xv3fsm58phqmtb40000gn/T/arduino_build_966129/libraries/arduino-lorawan/arduino-lorawan.a" "/var/folders/53/f8mlg0xn1xv3fsm58phqmtb40000gn/T/arduino_build_966129/libraries/SPI/SPI.cpp.o" "/var/folders/53/f8mlg0xn1xv3fsm58phqmtb40000gn/T/arduino_build_966129/libraries/Adafruit_FRAM_I2C/Adafruit_FRAM_I2C.cpp.o" "/var/folders/53/f8mlg0xn1xv3fsm58phqmtb40000gn/T/arduino_build_966129/libraries/Wire/Wire.cpp.o" "/var/folders/53/f8mlg0xn1xv3fsm58phqmtb40000gn/T/arduino_build_966129/libraries/RTCZero/RTCZero.cpp.o" "/var/folders/53/f8mlg0xn1xv3fsm58phqmtb40000gn/T/arduino_build_966129/libraries/arduino-lmic/aes/lmic.c.o" "/var/folders/53/f8mlg0xn1xv3fsm58phqmtb40000gn/T/arduino_build_966129/libraries/arduino-lmic/aes/other.c.o" "/var/folders/53/f8mlg0xn1xv3fsm58phqmtb40000gn/T/arduino_build_966129/libraries/arduino-lmic/aes/ideetron/AES-128_V10.cpp.o" "/var/folders/53/f8mlg0xn1xv3fsm58phqmtb40000gn/T/arduino_build_966129/libraries/arduino-lmic/hal/hal.cpp.o" "/var/folders/53/f8mlg0xn1xv3fsm58phqmtb40000gn/T/arduino_build_966129/libraries/arduino-lmic/lmic/lmic.c.o" "/var/folders/53/f8mlg0xn1xv3fsm58phqmtb40000gn/T/arduino_build_966129/libraries/arduino-lmic/lmic/lmic_as923.c.o" "/var/folders/53/f8mlg0xn1xv3fsm58phqmtb40000gn/T/arduino_build_966129/libraries/arduino-lmic/lmic/lmic_au921.c.o" "/var/folders/53/f8mlg0xn1xv3fsm58phqmtb40000gn/T/arduino_build_966129/libraries/arduino-lmic/lmic/lmic_eu868.c.o" "/var/folders/53/f8mlg0xn1xv3fsm58phqmtb40000gn/T/arduino_build_966129/libraries/arduino-lmic/lmic/lmic_eu_like.c.o" "/var/folders/53/f8mlg0xn1xv3fsm58phqmtb40000gn/T/arduino_build_966129/libraries/arduino-lmic/lmic/lmic_in866.c.o" "/var/folders/53/f8mlg0xn1xv3fsm58phqmtb40000gn/T/arduino_build_966129/libraries/arduino-lmic/lmic/lmic_us915.c.o" "/var/folders/53/f8mlg0xn1xv3fsm58phqmtb40000gn/T/arduino_build_966129/libraries/arduino-lmic/lmic/lmic_us_like.c.o" "/var/folders/53/f8mlg0xn1xv3fsm58phqmtb40000gn/T/arduino_build_966129/libraries/arduino-lmic/lmic/oslmic.c.o" "/var/folders/53/f8mlg0xn1xv3fsm58phqmtb40000gn/T/arduino_build_966129/libraries/arduino-lmic/lmic/radio.c.o" "/var/folders/53/f8mlg0xn1xv3fsm58phqmtb40000gn/T/arduino_build_966129/core/variant.cpp.o" -Wl,--start-group "-L/Users/lafleur/Library/Arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Lib/GCC/"  -larm_cortexM0l_math "-L/Users/lafleur/Library/Arduino15/packages/adafruit/hardware/samd/1.2.1/variants/feather_m0" -lm "/var/folders/53/f8mlg0xn1xv3fsm58phqmtb40000gn/T/arduino_build_966129/../arduino_cache_94165/core/core_adafruit_samd_adafruit_feather_m0_aa7d1bd3ceed48075940c2eda06ca867.a" -Wl,--end-group
/var/folders/53/f8mlg0xn1xv3fsm58phqmtb40000gn/T/arduino_build_966129/sketch/catena_hello_lora.ino.cpp.o: In function `__static_initialization_and_destruction_0':
/Users/lafleur/Desktop/MySensor-Arduino-development/MCCI/catena_hello_lora/catena_hello_lora.ino:23: undefined reference to `McciCatena::CatenaFeatherM0::LoRaWAN::LoRaWAN()'
collect2: error: ld returned 1 exit status
Using library Catena-Arduino-Platform at version 0.11.0 in folder: /Users/lafleur/Desktop/MySensor-Arduino-development/MCCI/libraries/Catena-Arduino-Platform 
[... cut ...]
Using library Adafruit_FRAM_I2C at version 1.0.0 in folder: /Users/lafleur/Desktop/MySensor-Arduino-development/MCCI/libraries/Adafruit_FRAM_I2C 
Using library Wire at version 1.0 in folder: /Users/lafleur/Library/Arduino15/packages/adafruit/hardware/samd/1.2.1/libraries/Wire 
Using library RTCZero at version 1.5.0 in folder: /Users/lafleur/Desktop/MySensor-Arduino-development/MCCI/libraries/RTCZero 
Using library arduino-lmic at version 2.1.5 in folder: /Users/lafleur/Desktop/MySensor-Arduino-development/MCCI/libraries/arduino-lmic 
exit status 1
Error compiling for board Adafruit Feather M0.

~~~

trlafleur, 2018-07-28:
well, I got a bit further by loading YOUR BSP, now complaining about EEPROM.h  

whos eeprom.h are you using?? as there are a number of them...

terrillmoore, 2018-07-28:
We don't use EEPROM.h, so I'm not sure which sketch you're compiling. Not enough context.  A quick search of Catena-Sketches confirms that.

Please check the README.md for Catena4450m101_sensor, https://github.com/mcci-catena/Catena-Sketches/blob/master/catena4450m101_sensor/README.md.  Please confirm that you can build catena4450m101_sensor.ino; this will verify that everything is properly installed.  If you have a problem with EEPROM.h, please open a new ticket (as we've now drifted from the `Serial.dtr())` discussion, and specify:

- the sketch you're trying to compile
- the target board
- provide a full verbose compile log (including the compiles as well as the links.

Thanks,
-Terry

closed 2018-07-28 by terrillmoore