mcci-catena/Catena-Sketches #42: Water Depth Sensor

NickWaterton, 2018-08-15:
In the catena4450m102_waterlevel sketch, you refer to an RC601 water pressure transducer, which uses an I2C interface.

Do you have a link to this device, or a datasheet? I can't find a reference to it anywhere on the web.
I am using this sensor [https://www.bluerobotics.com/store/electronics/bar02-sensor-r1-rp/s](url) but it's quite expensive (although it is super accurate, and low power).

Always looking for alternatives!

terrillmoore, 2018-08-15:
We get it from a vendor in China. I wasn't involved in the procurement, but I think @svelmurugan92 can post a reference here.

svelmurugan92, 2018-08-16:
Hello Nick,

We used Dalian Rayco's RC600 series water pressure transducer. Special version RC606 has I2C interface.
https://rayco20150306.en.ec21.com/RC600_Liquid_Level_Transmitter--9414097_9438509.html 

You need to contact Rayco for special requirements, Cable length, Output interface type etc..

FYR, Below is the photo of sensor

[image]

[image]

Thank you!

Best Regards,
Velmurugan S

closed 2019-02-13 by terrillmoore