mcci-catena/Catena-Sketches #44: Catena 4612 - no usb mount after upload catena461x_test01 sketch

everhamme, 2019-01-30:
I just uploaded a successfully compiled catena461x_test01 sketch and after it said upload done the Catena 4612 dismounted from USB and then never came back. The red light on the unit is blinking pretty fast.  any thoughts on how to get it back?

terrillmoore, 2019-01-31:
Did you attach a jumper (boot to Vdd) and reset? Unless the board is broken, that will bring up the USB boot loader. BTW, it's easier for us to do tech support over at https://portal.mcci.com (more people at MCCI get notified, including the key people who work with the 4612 every day).

everhamme, 2019-01-31:
No not yet.  That's the kind of reset trick I was looking for.  Thanks I'll post there next time.

terrillmoore, 2019-02-13:
Closed here, opened as an MCCI product support issue.

closed 2019-02-13 by terrillmoore