mcci-catena/Catena-Sketches #69: BSEC library has been updated - Need update for catena4460_bsec_ulp.ino

dhineshkumarmcci, 2019-02-27:
Bosch has updated the library for BSEC and the latest version is v1.4.7.2. 

Sketch **catena4460_bsec_ulp.ino** has been written based on BSEC software v1.4.6.0, so the sketch has to be updated to work with BSEC v1.4.7.2. Please find the code part need to be updated below:

https://github.com/mcci-catena/Catena-Sketches/blob/01c3a841b3f9fbc82fcdded2f0023268bac770f3/catena4460_bsec_ulp/catena4460_bsec_ulp.ino#L318-L327

Here in the sensor list, Bosch have changed output (virtual) sensor `BSEC_OUTPUT_IAQ_ESTIMATE` to `BSEC_OUTPUT_IAQ` in their library **bsec_datatypes.h** under "BSEC_1.4.7.2_Generic_Release_20190122\Arduino\BSEC\src\inc\". Also they have added three other output (virtual) sensors in their library:

```
BSEC_OUTPUT_STATIC_IAQ = 2,                             /*!< Unscaled indoor-air-quality estimate */ 
BSEC_OUTPUT_CO2_EQUIVALENT = 3,                         /*!< co2 equivalent estimate [ppm] */   
BSEC_OUTPUT_BREATH_VOC_EQUIVALENT = 4,                  /*!< breath VOC concentration estimate [ppm] */
``` 

Link for BSEC software: https://www.bosch-sensortec.com/bst/products/all_products/bsec#

dhineshkumarmcci, 2019-05-07:
@terrillmoore We have modified the **catena4460_bsec_ulp.ino** to work with the latest version of BSEC library v1.4.7.3. We see that the library has been written to support data either in int or float by handling **BME680_FLOAT_POINT_COMPENSATION** macro as per user's requirement.

As we calculate data as a float value, we have to un-comment **BME680_FLOAT_POINT_COMPENSATION** and it is added as a step in **README.md**.

With the long run, we see the IAQ accuracy remains to be '1' always. For your reference, we have added the log of few transmissions below:
```
-------------------------------------------------------------------------------
This is the catena4460_bsec_ulp program V0.2.1.
Target network: The Things Network / in866
Enter 'help' for a list of commands.
(remember to select 'Line Ending: Newline' at the bottom of the monitor window.)
--------------------------------------------------------------------------------

LoRaWAN init: succeeded
EUI64: 00-02-cc-01-00-00-00-9e
Platform Flags:  0x00009c09
Operating Flags:  0x00000001

BSEC library version 1.4.7.3
Got state from FRAM:
00000000  03 07 04 01 3d 00 00 00  00 00 00 00 73 00 00 00  ....=.......s...
00000010  2d 00 01 20 00 00 00 00  00 00 00 00 00 00 00 00  -.. ............
00000020  00 00 00 00 00 00 00 00  ff ff 02 02 00 00 00 00  ................
00000030  00 00 00 00 00 00 00 00  00 00 00 00 02 0c 00 02  ................
00000040  20 00 00 c8 41 00 00 c8  41 10 00 03 20 00 00 00   ...A...A... ...
00000050  00 00 00 00 00 00 00 00  00 16 00 05 20 00 00 00  ............ ...
00000060  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 0c  ................
00000070  00 09 20 ff ff ff ff ff  ff ff 7f 08 00 0a 20 00  .. ........... .
00000080  00 00 00 00 00 00 00 5a  fc 00 00                 .......Z...
6177, 32.06, 100583.92, 70.46, 964722.75, 25.00, 25.00, 500.00, 0.50, 6.67, 0.00, 0, 0, 0, 0, 0, 0, 32.06, 70.46
vBat:    4344 mV
BH1750:  9 lux
EV_TXSTART
EV_TXCOMPLETE
using light sleep
306171, 31.95, 100586.48, 70.65, 907244.44, 25.00, 25.00, 500.00, 0.50, 6.65, 0.00, 0, 0, 0, 0, 0, 0, 31.95, 70.65
vBat:    4357 mV
BME680:  T=32 P=100586 RH=71
BH1750:  11 lux
BME680:   Gas-Resistance=907244 IAQ=25
EV_TXSTART
EV_TXCOMPLETE
606171, 31.98, 100585.75, 70.78, 899328.88, 25.00, 25.00, 500.00, 0.50, 6.64, 0.00, 0, 0, 0, 0, 0, 0, 31.98, 70.78
vBat:    4344 mV
BME680:  T=32 P=100586 RH=71
BH1750:  15 lux
BME680:   Gas-Resistance=899329 IAQ=25
EV_TXSTART
EV_TXCOMPLETE
906171, 31.99, 100583.28, 70.59, 895123.56, 25.00, 25.00, 500.00, 0.50, 6.64, 0.00, 0, 0, 0, 0, 0, 0, 31.99, 70.60
vBat:    4344 mV
BME680:  T=32 P=100583 RH=71
BH1750:  16 lux
BME680:   Gas-Resistance=895124 IAQ=25
EV_TXSTART
EV_TXCOMPLETE
1206172, 32.08, 100574.11, 70.43, 896921.06, 25.00, 25.00, 500.00, 0.50, 6.64, 0.00, 1, 1, 1, 1, 0, 1, 32.08, 70.43
vBat:    4350 mV
BME680:  T=32 P=100574 RH=70
BH1750:  16 lux
BME680:   Gas-Resistance=896921 IAQ=25
EV_TXSTART
EV_TXCOMPLETE
1506172, 31.98, 100576.33, 70.24, 899328.88, 27.46, 26.07, 504.30, 0.51, 6.64, 1.09, 1, 1, 1, 1, 0, 1, 31.98, 70.24
1806171, 31.88, 100576.48, 70.54, 903573.88, 25.00, 25.00, 500.00, 0.50, 6.64, 0.00, 1, 1, 1, 1, 0, 1, 31.88, 70.54
vBat:    4344 mV
BME680:  T=32 P=100576 RH=71
BH1750:  16 lux
BME680:   Gas-Resistance=903574 IAQ=25
EV_TXSTART
EV_TXCOMPLETE
2106171, 31.83, 100564.17, 70.58, 907859.06, 25.00, 25.00, 500.00, 0.50, 6.65, 0.00, 1, 1, 1, 1, 0, 1, 31.83, 70.58
vBat:    4344 mV
BME680:  T=32 P=100564 RH=71
BH1750:  15 lux
BME680:   Gas-Resistance=907859 IAQ=25
EV_TXSTART
EV_TXCOMPLETE
```

terrillmoore, 2019-05-07:
See comments at https://github.com/mcci-catena/Catena-Arduino-Platform/issues/170 -- more work to be done.

dhineshkumarmcci, 2019-05-07:
@terrillmoore I have updated the sketch as per your comment.

terrillmoore, 2019-05-20:
Please submit a pull request.

commit 721f99e7 (2019-05-07): Fix #69: Removed the bsec parameters that are not transmitted
commit 914276a7 (2019-05-20): Fix #69: Updated the sketch to support latest BSEC library
commit 1354877a (2019-05-20): Fix #69: Updated the sketch to support latest BSEC library
commit 7b9d701e (2019-05-21): Fix #69: Updated the sketch to support latest BSEC library
closed 2019-05-22 by terrillmoore