mcci-catena/Catena-Sketches #116: catena4612_simple needs "Adafruit_Unified_Sensor"

sunburntcat, 2020-07-12:
The Adafruit_BME280_Library seems to require the Adafruit Unified Sensor library. Installing it manually takes care of the error.

terrillmoore, 2020-07-12:
Thanks, sorry. Actually, I should rename these libraries because we've diverged from upstream. But first things first.

terrillmoore, 2020-07-12:
Hi team (@dhineshkumarmcci @svelmurugan92 @sjpark-mcci), can you please update the .dat file so that git-boot.sh works properly, and then file a PR? Please do a horizontal scan. Since the name of the library is ambiguous (ours or Adafruits?) and we want to make sure customers get the right repo, we should then start the somewhat lengthy/time-consuming process of renaming the remaining libraries to avoid ambiguity. We can discuss that process offline.

dhineshkumarmcci, 2020-07-12:
Hi @terrillmoore , will do the needed changes and will submit a pull-request. Thanks!

commit 3bc735df (2020-07-13): Merge pull request #117 from mcci-catena/issue116

Fix #116: updated .dat file in catena4612_simple for the repo Adafruit_Sensor
closed 2020-07-13 by terrillmoore